xArm-Developer/xarm_ros2 #117: Move Robot to Initial position while (xbox controlling - moveit_servo)

lu4k87, 2025-02-19:
Hello I hope for help again.

I execute the following launch file:

**ros2 launch xarm_moveit_servo lite6_moveit_servo_realmove.launch.py robot_ip:=192.168.1.123 joystick_type:=1**

The control with the gamepad (xbox controller) works very well so far. i would now like to implement the following:

when I press the “Y” -> Y-BTN button on the xbox controller, the robot should move to the initial position.

I assume that I should use the:

**[xarm_msgs::srv::MoveCartesian]** (https://github.com/xArm-Developer/xarm_ros2/blob/master/xarm_msgs/srv/MoveCartesian.srv)

    **xarm_api->service: set_position
    xarm_api->service: set_tool_position
    xarm_api->service: set_position_aa
    xarm_api->service: set_servo_cartesian
    xarm_api->service: set_servo_cartesian_aa**

service call or should it be a different service? (do i have to switch/change the robot_mode or robot_state for this? don't quite understand what exactly i have to do...? 

I've been trying to find a solution for several days using chatgpt, but I keep running into problems and can't get it to work.
Please help :)

lu4k87, 2025-02-19:
### xarm_joystick_input.cpp

`**/* Copyright 2021 UFACTORY Inc. All Rights Reserved.
 *
 * Software License Agreement (BSD License)
 *
 * Author: Vinman <vinman.cub@gmail.com>
 ============================================================================*/

 #include "xarm_moveit_servo/xarm_joystick_input.h"
 #include "rclcpp/rclcpp.hpp"                        // -> Service Calls des Vakuum-Greifers
 #include "std_srvs/srv/set_bool.hpp"                // -> Statusaenderung des Vakuum-Greifers
 #include "xarm_msgs/srv/move_cartesian.hpp"         // -> move_position / cartesian ???
 
 
 
 namespace xarm_moveit_servo
 {
 
     // Joystick-Typen
     enum JOYSTICK_TYPE
     {
         JOYSTICK_xbox = 1 // Xbox One Elite 2 Controller
     };
 
     // Achsen des Xbox One Elite 2 Controllers
     enum xbox_CONTROLLER_AXIS
     {
         xbox_LEFT_STICK_LR = 0,
         xbox_LEFT_STICK_FB = 1,
         xbox_LEFT_TRIGGER = 2,
         xbox_RIGHT_STICK_LR = 3,
         xbox_RIGHT_STICK_FB = 4,
         xbox_RIGHT_TRIGGER = 5,
         xbox_CROSS_KEY_LR = 6,
         xbox_CROSS_KEY_FB = 7
     };
 
     // Buttons des Xbox One Elite 2 Controllers
     enum xbox_CONTROLLER_BUTTON
     {
         xbox_BTN_A = 0,
         xbox_BTN_B = 1,
         xbox_BTN_X = 2,
         xbox_BTN_Y = 3,
         xbox_BTN_LB = 4,
         xbox_BTN_RB = 5,
         xbox_BTN_BACK = 6,
         xbox_BTN_START = 7,
         xbox_BTN_POWER = 8,
         xbox_BTN_STICK_LEFT = 9,
         xbox_BTN_STICK_RIGHT = 10
     };
 
 // =========================================== Konstruktor - JoyToServoPub =======================================================================================================
     JoyToServoPub::JoyToServoPub(const rclcpp::NodeOptions &options)
         : Node("joy_to_twist_publisher", options),  // Node-Initialisierung
           dof_(7),                                  // Freiheitsgrade
           ros_queue_size_(10),
           joystick_type_(JOYSTICK_xbox), // Standard-Joystick-Typ
           initialized_status_(10),                  // Initialstatus
           joy_topic_("/joy_check"),                 // Joystick-Topic
           cartesian_command_in_topic_("/servo_server/delta_twist_cmds"),    // Kartesische Befehle
           joint_command_in_topic_("/servo_server/delta_joint_cmds"),        // Gelenk-Befehle
           robot_link_command_frame_("link_base"),
           ee_frame_name_("link_eef"),                                       // EE Frame
           planning_frame_("link_base"),                                     // Base Frame
           greifer_status_(false)                                            // Gripper-Status
     {
         // Parameter initialisieren
[... 200 lines cut ...]
 
         // Veroeffentliche Twist- oder Joint-Befehle =========================================================
         if (pub_twist)
         {
             _filter_twist_msg(twist_msg, 0.2);
             twist_msg->header.frame_id = planning_frame_;
             twist_msg->header.stamp = this->now();
             twist_pub_->publish(std::move(twist_msg));
         }
         else
         {
             joint_msg->header.stamp = this->now();
             joint_msg->header.frame_id = "joint";
             joint_pub_->publish(std::move(joint_msg));
         }//END: Veroeffentliche Twist- oder Joint-Befehle ====================================================
 
     }//END: Callback: joy_callback() fuer Joystick-Nachrichten ===============================================
     
 }//END: namespace xarm_moveit_servo ==========================================================================
 
 
 // Node registrieren
 #include <rclcpp_components/register_node_macro.hpp>
 RCLCPP_COMPONENTS_REGISTER_NODE(xarm_moveit_servo::JoyToServoPub)
 **`

lu4k87, 2025-02-20:
This is the Code for **xarm_joystick_input.h**:

#ifndef __XARM_JOY_STICK_INPUT_H__
#define __XARM_JOY_STICK_INPUT_H__

#include <rclcpp/rclcpp.hpp>
#include <sensor_msgs/msg/joy.hpp>
#include <geometry_msgs/msg/twist_stamped.hpp>
#include <control_msgs/msg/joint_jog.hpp>
#include <std_srvs/srv/trigger.hpp>
#include <moveit_msgs/msg/planning_scene.hpp>
#include <xarm_msgs/srv/call.hpp>               // <<---- Header-Datei fuer den Gripper-Service function
#include <xarm_msgs/srv/get_float32_list.hpp>   // <<---- Header-Datei (get) fuer z-position 

**#include <xarm_msgs/srv/set_int16.hpp>          // <<---- HINZUGEFUEGT: SetMode/SetState Service
#include <xarm_msgs/srv/move_cartesian.hpp>     // <<---- HINZUGEFUEGT: MoveCartesian (set_position) Service**

namespace xarm_moveit_servo
{

class JoyToServoPub : public rclcpp::Node
{
public:
    JoyToServoPub(const rclcpp::NodeOptions& options);
private:
    template <typename T>
    void _declare_or_get_param(T& output_value, const std::string& param_name, const T default_value = T{});
    bool _convert_gamepad_joy_to_cmd(
        const std::vector<float>& axes, const std::vector<int>& buttons,
        std::unique_ptr<geometry_msgs::msg::TwistStamped>& twist,
        std::unique_ptr<control_msgs::msg::JointJog>& joint);

    void _filter_twist_msg(std::unique_ptr<geometry_msgs::msg::TwistStamped>& twist, double val);
    void _joy_callback(const sensor_msgs::msg::Joy::SharedPtr msg);

    
    rclcpp::Subscription<sensor_msgs::msg::Joy>::SharedPtr joy_sub_;                // Subscriber   -> Xbox Controller eingaben
    rclcpp::Publisher<geometry_msgs::msg::TwistStamped>::SharedPtr twist_pub_;      // pubisher     -> msg::TwistStamped
    rclcpp::Publisher<control_msgs::msg::JointJog>::SharedPtr joint_pub_;           // pubisher     -> msg::JointJog
    rclcpp::Publisher<moveit_msgs::msg::PlanningScene>::SharedPtr collision_pub_;   // pubisher     -> msg::PlanningScene
    rclcpp::Client<std_srvs::srv::Trigger>::SharedPtr servo_start_client_;          // servo_start_client
	
    
//------------------------------------------------ HINZUGEFUEGT: --------------------------------------------------------------
// Deklaration der Gripper-Service-Clients fuer den Gripper-Service "Call"
    rclcpp::Client<xarm_msgs::srv::Call>::SharedPtr open_gripper_client_;
    rclcpp::Client<xarm_msgs::srv::Call>::SharedPtr close_gripper_client_;
// Deklaration der get Position-Clients 
    rclcpp::Client<xarm_msgs::srv::GetFloat32List>::SharedPtr get_position_client_;
 

 // ------------------ NEU: Clients für Mode/State und Stop/Start Servo, sowie MoveCartesian -------------------
 **rclcpp::Client<xarm_msgs::srv::SetInt16>::SharedPtr set_mode_client_;
 rclcpp::Client<xarm_msgs::srv::SetInt16>::SharedPtr set_state_client_;
 rclcpp::Client<std_srvs::srv::Trigger>::SharedPtr servo_stop_client_;
 rclcpp::Client<xarm_msgs::srv::MoveCartesian>::SharedPtr set_position_client_;**
//----------------------------------------------------------------------------------------------------------------------------
    int dof_;
    int ros_queue_size_;
    int joystick_type_;
    int initialized_status_;
    std::string joy_topic_;
    std::string cartesian_command_in_topic_;
    std::string joint_command_in_topic_;
    std::string robot_link_command_frame_;
    std::string ee_frame_name_;             // Endeffector Frame
    std::string planning_frame_;
    bool greifer_status_;                   // <-- Status des Greifers (open/closed)    
    float current_z_;                       // <-- aktuelle z-position
    
};
}

#endif // __XARM_JOY_STICK_INPUT_H__

lu4k87, 2025-02-20:
Compiling works (colcon build)!

after source and execute i got this message from Terminal:

[component_container-5] [INFO] [1740055918.919011258] [joy_to_servo_node]: Y-Button pressed: Wechsel zu Position Control, set_position aufrufen, dann zurück in SERVO-Modus
[component_container-5] [WARN] [1740055923.919303301] [joy_to_servo_node]: stop_servo service call timed out
[ros2_control_node-3] [set_state], xArm is not ready to move
[ros2_control_node-3] [WARN] [1740055923.926739464] [UFACTORY.RobotHW]: [192.168.1.175] Robot State detected! State: 4
[ros2_control_node-3] [WARN] [1740055924.928954681] [UFACTORY.RobotHW]: [192.168.1.175] Robot State detected! State: 5
[component_container-5] [WARN] [1740055928.919815018] [joy_to_servo_node]: set_mode(0) timed out

penglongxiang, 2025-02-21:
Hi, the `moveit_servo` node should operate the robot in **Mode 1** (external planner). However, the execution of service type `MoveCartesian` requires **Mode 0** (integrated planner). So although not recommended, proper mode switch should be done if you want to mix the control. Remember to set state (0) after mode switch, please **update to the latest code** for more safety logic, do your safety assessment and be careful for unexpected movement when you mix the control.

lu4k87, 2025-02-25:
Hello, thank you for your answer.

so again, i would like to launch while using moveit servo (the following command was called: 

**ros2 launch xarm_moveit_servo lite6_moveit_servo_realmove.launch.py robot_ip:=192.168.1.123 joystick_type:=1**

everything works as desired. if i understand correctly, the robot works in mode:1. 
Now I want to perform a mode switch **while the moveit_servo** is running.

For example, if I press the Y-Button(xbox gamepad), the mode switch (mode:1 -> mode:0) should be executed and the service call: 

**- set_position as (MoveCartesian message)** 

should be called. 

As soon as the robot (endeffector) has reached the position, I want to switch back to mode:1 to send the trajectory messages as before.

What exactly does the code for this look like? (C++)
what is the sequence to perform the mode switch?

I use the **xarm_joystick_input.cpp** from the repository.

lu4k87, 2025-02-26:
perhaps one more question first.

I want to call a service call (when I press the Y button) that moves the end effector to the desired position. 

the position should be specified in the format: (x,y,z,roll,pitch,yaw). i want to pass the coordinates (as in the Ufactory Studio software) to the service call.

Which service call is best suited for this?
**- set_position / _aa?**
**- set_servo_cartesian / _aa?**
**- another service call?**

penglongxiang, 2025-02-27:
Please use `set_position()` in mode 0 if the target orientation is expressed in RPY.

lu4k87, 2025-02-27:
Okay, I've now got it (halfway) up and running.

Summarized again:
1. **>> ros2 launch xarm_moveit_servo lite6_moveit_servo_realmove.launch.py robot_ip:=192.168.1.123 joystick_type:=1 report_type:=dev**

2. the movement of arm (end effector) runs as it should, everything is great (gamepad inputs)

3. when i press: Y-BTN -> should start the **set_position - process**

3.1 switch **mode** & **state** to “**0**” (in**position mode** -> that was the idea and it works!) 
(I see this in ufactory studio and in the terminal) 

3.2 Service Call: **set_position()** 

3.4 switch mode to “1” & state to “0” so switch back again (in **servoj mode**)

the Service Calls work and the endeffector also moves to the desired position!!!  

5. after the desired position has been reached, the robot (i.e. the endeffector) moves back to the previous position at high speed. (BUT WHY??) 

6. i can then move the robot again with the axes of the gamepad controller (mode:1 state:0)

To summarize:
the set_mode() and set_state() work, also the set_position() works! 

but why does the arm move back again? why doesn't it stay at the position after **set_position + switch back to mode:1**?

if i execute the set_mode, set_switch, set_position procedure in the **terminal** everything works as desired! So it must have something to do with the script (code)?

Thanks for the help... 

TERMINAL OUTPUT:
[INFO] [launch]: All log files can be found below /home/labadmin/.ros/log/2025-02-27-20-00-39-811102-labadmin-MacPro-74537
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rviz2-1]: process started with pid [74541]
[INFO] [joint_state_publisher-2]: process started with pid [74543]
[INFO] [ros2_control_node-3]: process started with pid [74545]
[INFO] [spawner-4]: process started with pid [74547]
[rviz2-1] Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
[ros2_control_node-3] [WARN] [1740682840.386059701] [controller_manager]: [Deprecated] Passing the robot description parameter directly to the control_manager node is deprecated. Use '~/robot_description' topic from 'robot_state_publisher' instead.
[ros2_control_node-3] [INFO] [1740682840.386410853] [resource_manager]: Loading hardware 'uf_robot_hardware/UFRobotSystemHardware' 
[ros2_control_node-3] [INFO] [1740682840.402319766] [resource_manager]: Initialize hardware 'uf_robot_hardware/UFRobotSystemHardware' 
[ros2_control_node-3] [INFO] [1740682840.417685759] [UFACTORY.RobotHW]: [192.168.1.175] namespace: /
[ros2_control_node-3] [INFO] [1740682840.417716123] [UFACTORY.RobotHW]: [192.168.1.175] robot_type: lite, hw_ns: ufactory, prefix: , report_type: dev
[ros2_control_node-3] [INFO] [1740682840.417980846] [UFACTORY.RobotHW]: [192.168.1.175] dof: 6, velocity_control: 0, add_gripper: 0, add_bio_gripper: 0, baud_checkset: 1, default_gripper_baud: 2000000
[ros2_control_node-3] [INFO] [1740682840.418245065] [ufactory_driver]: robot_ip=192.168.1.175, report_type=dev, dof=6
[ros2_control_node-3] [INFO] [1740682840.418291701] [ufactory_driver]: baud_checkset: 1, default_gripper_baud: 2000000
[ros2_control_node-3] SDK_VERSION: 1.14.4
[ros2_control_node-3] Tcp control connection successful
[ros2_control_node-3] ROBOT_IP: 192.168.1.175, VERSION: v2.5.5, PROTOCOL: V1, DETAIL: 6,9,LI1006,DL1000,v2.5.5, TYPE1300: [0, 0]
[ros2_control_node-3] Tcp Report Rich connection successful
[ros2_control_node-3] change protocol identifier to 3
[ros2_control_node-3] report_data_size: 574, size_is_not_confirm: 0
[ros2_control_node-3] Tcp Report develop connection successful
[ros2_control_node-3] [INFO] [1740682840.746083687] [ufactory_driver]: [TCP STATUS] CONTROL: 1, REPORT: 1
[ros2_control_node-3] report_data_size: 135, size_is_not_confirm: 0
[ros2_control_node-3] [INFO] [1740682840.891002163] [ufactory_driver]: gripper_speed: 2000, gripper_max_pos: 850, gripper_frequency : 10, gripper_threshold: 3, gripper_threshold_times: 10
[ros2_control_node-3] [INFO] [1740682840.900388450] [ufactory_driver]: bio_gripper_speed: 2000, bio_gripper_max_pos: 130, bio_gripper_min_pos: 50, bio_gripper_frequency : 10, bio_gripper_threshold: 3, bio_gripper_threshold_times: 10
[ros2_control_node-3] [INFO] [1740682840.903301081] [UFACTORY.RobotHW]: [192.168.1.175] System Sucessfully configured!
[ros2_control_node-3] [INFO] [1740682840.903341637] [resource_manager]: Successful initialization of hardware 'uf_robot_hardware/UFRobotSystemHardware'
[ros2_control_node-3] [INFO] [1740682840.903542571] [resource_manager]: 'configure' hardware 'uf_robot_hardware/UFRobotSystemHardware' 
[ros2_control_node-3] [INFO] [1740682840.903556098] [resource_manager]: Successful 'configure' of hardware 'uf_robot_hardware/UFRobotSystemHardware'
[ros2_control_node-3] [INFO] [1740682840.903565648] [resource_manager]: 'activate' hardware 'uf_robot_hardware/UFRobotSystemHardware' 
[ros2_control_node-3] [set_state], xArm is ready to move
[spawner-4] [INFO] [1740682840.943157930] [spawner_lite6_traj_controller]: waiting for service /controller_manager/list_controllers to become available...
[ros2_control_node-3] [INFO] [1740682840.943363104] [UFACTORY.RobotHW]: [192.168.1.175] System Sucessfully started!
[ros2_control_node-3] [INFO] [1740682840.943392316] [resource_manager]: Successful 'activate' of hardware 'uf_robot_hardware/UFRobotSystemHardware'
[rviz2-1] [INFO] [1740682840.958873268] [rviz2]: Stereo is NOT SUPPORTED
[... cut ...]
    std::string joy_topic_;          // Topic für Joystick-Eingaben
    std::string cartesian_command_in_topic_;
    std::string joint_command_in_topic_;
    std::string robot_link_command_frame_;
    std::string ee_frame_name_;      // Endeffektor Frame
    std::string planning_frame_;     // Planungs-Frame
    bool greifer_status_;            // Status des Greifers (open/closed)    
    float current_z_;                // Aktuelle z-Position
};

} // namespace xarm_moveit_servo

#endif // __XARM_JOY_STICK_INPUT_H__

penglongxiang, 2025-02-28:
@lu4k87 Please update to the latest version of code ([5e3f1df](https://github.com/xArm-Developer/xarm_ros2/commit/5e3f1df7ec4a8fffdf559bd46842bc8b0029da82) updated 2 weeks ago), we added additional logic to handle the situation when switching control between ros2 controller and xArm native firmware, the ROS2 controller should be deactivated explicitly when switching to mode 0, and then activated again when mode is back to 1, or the ros controller running background would thought the position of robot is not changed. Please update and have a test with your current logic.

lu4k87, 2025-02-28:
hello thanks for the answer!

i have done that, unfortunately without success. the problem persists...

the position in (mode:0) is not updated for (mode:1). because as soon as i switch back to mode:1, the arm moves back to the previous position (as described above)

however, I have now found out that I can probably reach my target via the **MoveGroup Interface**:

`bool moveit::planning_interface::MoveGroup::setPositionTarget (double x, double y, double z, const std::string & end_effector_link = “”)`
(http://docs.ros.org/en/jade/api/moveit_ros_planning_interface/html/classmoveit_1_1planning__interface_1_1MoveGroup.html#a5ac7f146deeee5733d9757a16f52463a)

I will try this out over the next few days and then give you feedback.

lu4k87, 2025-03-04:
can you please tell me which lines i have to change in the code so that it finally works?

penglongxiang, 2025-03-05:
Hi sorry we have no spare time currently to examine your code. But I confirm with the latest code of xarm_ros2 humble branch, launching 
`ros2 launch xarm_moveit_servo xarm_moveit_servo_realmove.launch.py dof:=7 robot_ip:=192.168.1.127`
and control with keyboard by:
`ros2 run xarm_moveit_servo xarm_keyboard_input`

Then doing the mode/state switch and calling `/xarm/move_gohome` service, then mode/switch back again with the corresponding services in another terminal. The robot can still be controlled by moveit servo through keyboard with no sudden jump on the real robot.

lu4k87, 2025-04-09:
Hello, 
I'm still having this problem and wanted to ask if anyone here can help me?

how should i proceed if i create a new node for my project? what makes the most technical sense?

maybe i have to update the joint_states in the planning scene after the mode switch? (but how exactly?) which header files do I have to include and what is the code in the script?

or perhaps use the linear movements from the script in a function with the help of a for loop until the position (defined target position) has been reached?

penglongxiang, 2025-06-30:
@lu4k87 Thank you for the detailed feedback and glad to know your problem has been solved.