xArm-Developer/xarm_ros2 #118: Failed to configure controller

edompedro, 2025-02-25:
Hi, I'm trying to run the base moveit simulation with xarm6, but, unfortunately, I can't load the controller. It looks like the URDF links for the robot description or something like that it isn't load or connected with the controller. All my files are the same as this repository, I'm using ROS2 Humble. The error I'm getting is describe bellow: 

[ros2_control_node-3] [ERROR] [1740446301.575442349] [controller_manager]: After configuring, controller 'joint_state_broadcaster' is in state 'unconfigured' , expected inactive.
[spawner-2] [ERROR] [1740446301.575723622] [spawner_joint_state_broadcaster]: Failed to configure controller
[spawner-4] [INFO] [1740446301.580118148] [spawner_xarm6_traj_controller]: Loaded xarm6_traj_controller
[ros2_control_node-3] [INFO] [1740446301.580606672] [controller_manager]: Configuring controller 'xarm6_traj_controller'
[ros2_control_node-3] [INFO] [1740446301.580722921] [xarm6_traj_controller]: No specific joint names are used for command interfaces. Using 'joints' parameter.
[ros2_control_node-3] [INFO] [1740446301.580734290] [xarm6_traj_controller]: Command interfaces are [position velocity] and state interfaces are [position velocity].

[image]

penglongxiang, 2025-02-27:
Hi, we have no problem running with ROS2 humble. Could you provide the exact command and arguments to reproduce this issue? It looks like something wrong with the controller manager, have you checked that everything is properly installed on your PC?

edompedro, 2025-02-27:
Hi, thanks for your quick response. The problem it was on my notebook and I did not realized that early. I moved back and reinstall the package in my desktop and all works fine. Probably the error was when I configured the MoveIt previously. Sorry to spend your time with that.

closed 2025-02-27 by edompedro