xArm-Developer/xarm_ros2 #120: Joint start state tolerances fail after collision

BestIlyaPetrov, 2025-03-19:
**My environment**
ROS Distro: Humble
OS Version: Ubuntu 22.04

I launch rviz using either **ros2 launch xarm_planner xarm6_planner_realmove.launch.py robot_ip:=192.168.1.XXX** or **ros2 launch xarm_moveit_config xarm6_moveit_realmove.launch.py robot_ip:=192.168.1.XXX**

1. After initial launch, I can create a new end goal for my xarm6, then plan&execute the trajectory. If the robot collides (i hold it by hand) while executing the trajectory, the state tolerances fail -> the trajectory execution fails. This is expected.

2. I then use the Ufactory App to reset the collision error. After the error is reset, I can move the robot from the Ufactory App - no problem.
3. However, if i then try to create a new end goal and execute the trajectory, the robot says the following: (this happens upon pressing the "execute" trajectory button in the MoveIt panel in Rviz)

[move_group-3] [INFO] [1742419484.861351526] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: sending trajectory to xarm6_traj_controller
[ros2_control_node-5] [INFO] [1742419484.861670975] [xarm6_traj_controller]: Received new action goal
[ros2_control_node-5] [INFO] [1742419484.861731821] [xarm6_traj_controller]: Accepted new action goal
[move_group-3] [INFO] [1742419484.861817524] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: xarm6_traj_controller started execution
[move_group-3] [INFO] [1742419484.861828905] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Goal request accepted!
[ros2_control_node-5] [ERROR] [1742419486.455718718] [tolerances]: State tolerances failed for joint 0:
[ros2_control_node-5] [ERROR] [1742419486.455737651] [tolerances]: Position Error: -0.017611, Position Tolerance: 0.010000
[ros2_control_node-5] [ERROR] [1742419486.455765137] [tolerances]: State tolerances failed for joint 1:
[ros2_control_node-5] [ERROR] [1742419486.455766736] [tolerances]: Position Error: 0.049781, Position Tolerance: 0.010000
[ros2_control_node-5] [ERROR] [1742419486.455768545] [tolerances]: State tolerances failed for joint 2:
[ros2_control_node-5] [ERROR] [1742419486.455769727] [tolerances]: Position Error: 0.057993, Position Tolerance: 0.010000
[ros2_control_node-5] [ERROR] [1742419486.455771306] [tolerances]: State tolerances failed for joint 3:
[ros2_control_node-5] [ERROR] [1742419486.455772460] [tolerances]: Position Error: 0.028354, Position Tolerance: 0.010000
[ros2_control_node-5] [ERROR] [1742419486.455774073] [tolerances]: State tolerances failed for joint 4:
[ros2_control_node-5] [ERROR] [1742419486.455775272] [tolerances]: Position Error: 0.166250, Position Tolerance: 0.010000
[ros2_control_node-5] [ERROR] [1742419486.455776923] [tolerances]: State tolerances failed for joint 5:
[ros2_control_node-5] [ERROR] [1742419486.455778083] [tolerances]: Position Error: -0.031803, Position Tolerance: 0.010000
[ros2_control_node-5] [WARN] [1742419486.455783813] [xarm6_traj_controller]: Aborted due to goal_time_tolerance exceeding by 0.500016 seconds
[move_group-3] [WARN] [1742419486.462124781] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Controller 'xarm6_traj_controller' failed with error GOAL_TOLERANCE_VIOLATED: Aborted due to goal_time_tolerance exceeding by 0.500016 seconds
[move_group-3] [WARN] [1742419486.462252184] [moveit_ros.trajectory_execution_manager]: Controller handle xarm6_traj_controller reports status ABORTED
[move_group-3] [INFO] [1742419486.462274339] [moveit_ros.trajectory_execution_manager]: Completed trajectory execution with status ABORTED ...
[move_group-3] [INFO] [1742419486.462414441] [moveit_move_group_default_capabilities.execute_trajectory_action_capability]: Execution completed: ABORTED
[rviz2-2] [INFO] [1742419486.462747528] [move_group_interface]: Execute request aborted
[rviz2-2] [ERROR] [1742419486.462965068] [move_group_interface]: MoveGroupInterface::execute() failed or timeout reached

------
The path gets planned just fine, but the execution fails because the start state becomes out of sync. /joint_states continues publishing the joint values correctly, but the robot's controller seems to stop reading them after a collision or any other error happens.

**The only way to move the robot with MoveIt again, is to re-launch the xarm_planner pipeline (via ros2 launch xarm_planner xarm6_planner_realmove.launch.py robot_ip:=192.168.1.XXX).**

I have been trying to dig through the code and locate the core reason why the controller stops reading the latest joint values, but i haven't been able to find where this comes from. My goal is to be able to reset the robot state after the collision & continue executing trajectories without re-starting the whole 'xarm_planner' or 'xarm_moveit_config' launch pipeline.

Please help! I'm happy to share more information & help debug this.

penglongxiang, 2025-03-20:
Hi @BestIlyaPetrov, it may not be very clear in the doc that Moveit + ros control should operate in Mode 1 (refer [here](https://github.com/xArm-Developer/xarm_ros2/tree/humble/xarm_api#robot-mode-1)) . However if error code pops up, the system will automatically switch back to default Mode 0. 

To recover control in Moveit application without restarting it, you can switch back to Mode 1 and state 0 manually, by calling the following service **in order** in another terminal:
```shell
$ ros2 service call /xarm/set_mode xarm_msgs/srv/SetInt16 "{data: 1}"
$ ros2 service call /xarm/set_state xarm_msgs/srv/SetInt16 "{data: 0}"
```
You may use a script to do the switch for simplicity. Moreover, please **make sure you have updated to the latest code** of this repository, we have optimized the safety logic for this situation, or sudden jump may occur during the mode switch with ros controller running in background.

BestIlyaPetrov, 2025-03-20:
Thank you so much! This worked like a charm.

Also, yes, I saw a small jump in the joints when I switched modes. I'll pull the latest updates. Thanks again.

closed 2025-03-20 by BestIlyaPetrov