xArm-Developer/xarm_ros2 #121: [!Bug] DualMoveItConfigsBuilder lite_gripper add_gripper issue

pala5, 2025-03-21:
Hi, I am experiencing some issues when I try to visualize the dual lite6 moveit configuration adding the **lite gripper**, enabling the _add_gripper_ boolean flag. I know that the lite gripper is not incorporated within the ros2_control framework, mainly because it is activated through ROS services, ON/OFF control. But in my current application it will be needed to take into consideration its physical geometry to avoid collisions.

## Error reprodution
First of all, you need to include in the [dual_lite6_moveit_fake.launch.py](https://github.com/xArm-Developer/xarm_ros2/blob/humble/xarm_moveit_config/launch/dual_lite6_moveit_fake.launch.py) file the **add_gripper** parameter set to true into the launch arguments:
```    
robot_moveit_fake_launch = IncludeLaunchDescription(
        PythonLaunchDescriptionSource(PathJoinSubstitution([FindPackageShare('xarm_moveit_config'), 'launch', '_dual_robot_moveit_fake.launch.py'])),
        launch_arguments={
            'dof': '6',
            'robot_type': 'lite',
            'hw_ns': hw_ns,
            'no_gui_ctrl': 'false',
            'add_gripper': 'true',
        }.items(),
```
Then you have to execute the respective launch file:
```
ros2 launch xarm_moveit_config dual_lite6_moveit_fake.launch.py
```
Outcome
```
Caught exception in launch (see debug for traceback): "File ${workspace path}/install/share/xarm_moveit_config/config/lite_gripper/fake_controllers.yaml doesn't exist"
```
## Proposed Solution
Modify in the [uf_ros_lib/uf_ros_lib/moveit_configs_build.py](https://github.com/xArm-Developer/xarm_ros2/blob/humble/uf_ros_lib/uf_ros_lib/moveit_configs_builder.py) file the function **trajectory_execution** from the class **DualMoveItConfigsBuilder** to include into the if statement of both lines 1293 and 1310 the _self.__robot_type_1 != 'lite'_ and _self.__robot_type_2 != 'lite'_, respectively.
### Current version
```
if self.__add_gripper_X in ('True', 'true'):
```
### Solution
```
if self.__robot_type_X != 'lite' and self.__add_gripper_X in ('True', 'true'):
```

vimior, 2025-03-26:
@pala5 
Thank you for your support and suggestions. This is a bug and has been fixed now.

pala5, 2025-04-04:
@vimior 
Thanks!

closed 2025-04-04 by pala5