xArm-Developer/xarm_ros2 #123: MoveIt Servo Pose Command

gauthammsam57, 2025-04-15:
Hello there,

I am currently exploring MoveIt Servo using the Lite6 model. I was able to figure out how to use Twist messages to move the robot, essentially using velocity control.

I noticed that there is a custom set_servo_cartesian service provided, which seems to offer functionality similar to what I’m looking for. However, I couldn’t get it to work with the Lite6.

That said, I would still prefer to find a way to achieve this using MoveIt, as it would be much easier to integrate into our existing system.

Is there a way to perform real-time pose control with MoveIt?

Thanks for the response in advance.

penglongxiang, 2025-04-24:
Hi, Moveit Servo will match your need for real-time pose control with Moveit, we provided demos for `moveit_servo` control in simulation and real world, have you tried it? Refer [here](https://github.com/xArm-Developer/xarm_ros2?tab=readme-ov-file#59-xarm_moveit_servo)

gauthammsam57, 2025-04-24:
Thanks for the response @penglongxiang.

I did refer to the moveit_servo example. But looking at both, they are velocity based. I was wondering if there are any way to have moveit_servo in position based. Where I will sent: x, y, z and orientation. Then it will move to that point in real time.

penglongxiang, 2025-04-25:
Hi @gauthammsam57, I think the default control interface is already position instead of velocity. But I am not sure if I get this correct, is it because moveit_servo uses Jacobian based calculation that you would not prefer (maybe sometimes the accuracy is not high enough)? If this is what you mean, then I am currently not aware of other servoing method under moveit framework.

We do provide `set_servo_cartesian` (roll-pitch-yaw representation) and `set_servo_cartesian_aa` (axis-angle representation) API in mode 1 to servo control the robot arm, however you would have to generate smooth trajectory in real-time by yourself. 

If motion planning is not possible on your side, there is another option you can try, using `set_position()` under **Mode 7** (On-line trajectory planning mode), check the example [here](https://github.com/xArm-Developer/xArm-Python-SDK/blob/master/example/wrapper/common/1010-cartesian_online_trajectory_planning.py), it will eliminate the requirement for self planning but still achieve dynamic control, but there may be some delay under this mode. You may have a try to see if this is suitable for your case.

Shahar-AstroScale, 2025-04-28:
follow up question regarding xarm moveit servo, seems like you implemented just the human controlled functionalities (teleop keyboard / joystick), and not pose tracking servoing, which is a realtime control solution provided by moveit, is there an intention to implement not by using the xarm api but using moveit?

refer to the following video as reference:
https://www.youtube.com/watch?v=t2V4llsSxW4&t=1s

thanks.

penglongxiang, 2025-04-29:
Please be more specific on your intended function. 

For me the pose tracking/visual servoing have the similar logic with the joystick control, which is updating the `delta_twist_cmds` regularly from upper node to the moveit_servo  (`servo_server`) node and it tries to compute the next target joint command (through jacobian based IK and collision checking) in soft real-time. We may not implement all the cases for what moveit servo can support.

gauthammsam57, 2025-05-05:
I wanted the same functionality as what said: https://github.com/xArm-Developer/xarm_ros2/issues/123#issuecomment-2834622284. 

Thanks for the response @penglongxiang.

penglongxiang, 2025-05-08:
@gauthammsam57, this request is not very clear. The perception part is really application specific, but the driving principle is very similar. Developers shall comprehend and study the teleop demo, understand the message passing behind the scene, and integrate moveit_servo function with their perception device SDK by themselves, thanks.