xArm-Developer/xarm_ros2 #124: [BUG] Dual xarm fake controller bug

BestIlyaPetrov, 2025-04-17:
Environment:
ROS Distro: Humble
OS Version: Ubuntu 22.04

Hi, 

I tried launching the dual arm simulator and stumbled upon the following issue. The two arms get loaded and configured in rviz just fine, both the L_xarm6_traj_controller and the R_xarm6_traj_controller get loaded successfully. However, when i set a new goal on of the two robots and click "plan", i get the following error:

[rviz2-2] [INFO] [1744931865.671165563] [move_group_interface]: MoveGroup action client/server ready
[move_group-5] [INFO] [1744931865.671603739] [moveit_move_group_default_capabilities.move_action_capability]: Received request
[move_group-5] [INFO] [1744931865.671748351] [moveit_move_group_default_capabilities.move_action_capability]: executing..
[rviz2-2] [INFO] [1744931865.671764643] [move_group_interface]: Planning request accepted
[move_group-5] [INFO] [1744931865.676842757] [moveit_move_group_default_capabilities.move_action_capability]: Planning request received for MoveGroup action. Forwarding to planning pipeline.
[move_group-5] [INFO] [1744931865.676930314] [moveit_move_group_capabilities_base.move_group_capability]: Using planning pipeline 'ompl'
[move_group-5] [INFO] [1744931865.683954885] [moveit.ompl_planning.model_based_planning_context]: Planner configuration 'L_xarm6' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[move_group-5] [ERROR] [1744931865.692587793] [moveit_exceptions.exceptions]: Variable 'joint4' is not known to model 'DUAL_UF_ROBOT'
[move_group-5] Exception thrown.
[move_group-5] terminate called after throwing an instance of 'moveit::Exception'
[move_group-5]   what():  Variable 'joint4' is not known to model 'DUAL_UF_ROBOT'
[ERROR] [move_group-5]: process has died [pid 8307, exit code -6, cmd '/home/dellubuntu/ws_moveit2/install/moveit_ros_move_group/lib/moveit_ros_move_group/move_group --ros-args --params-file /tmp/launch_params_u6u37sg7 --params-file /tmp/launch_params_dkqnu8ai'].
[move_group-5] 
[rviz2-2] [INFO] [1744931908.623703522] [moveit_ros_robot_interaction.robot_interaction]: No active joints or end effectors found for group 'dual_arm'. Make sure that kinematics.yaml is loaded in this node's namespace.
[rviz2-2] [INFO] [1744931915.530210180] [move_group_interface]: MoveGroup action client/server not ready

--- 
To reproduce, launch via: "ros2 launch xarm_moveit_config dual_xarm6_moveit_fake.launch.py"

Seems like "L_" and "R_" prefixes don't get appended somewhere, but I'm struggling to locate where.

[image]

penglongxiang, 2025-04-24:
Hi, have you changed any configurations to the original code? I have checked on my side with humble branch, and it worked without problem. I notice there is a planning group called 'dual_arm' in your provided log which does not exist in original code.

BestIlyaPetrov, 2025-05-01:
I'm sorry. I forgot I edited a file in MoveIt a long time ago. That was the cause for this error. The dual simulator does work correctly. Thank you for checking.

closed 2025-05-01 by BestIlyaPetrov