xArm-Developer/xarm_ros2 #125: Automatic Handeye Camera Calibration and Reprojection Error

cemoke, 2025-04-23:
Hi, I saw you have a hand-eye camera calibration routine in ROS 1 https://github.com/xArm-Developer/xarm_ros/tree/master/xarm_vision

It'd be very useful to have it in ROS2 as well. Can you bring this implementation to ROS2?

Secondly, what is the expected reprojection error after the hand-eye calibration with uLite 6 and a Realsense camera (D415)?

Our error is around 1cm (measured by evaluate launch file of the easy_handeye2 package) and we are curious whether this much error is to be expected.

Thank you in advance!

cemoke, 2025-05-08:
Hello, at least can someone respond to what is the expected position error after the camera calibration?

penglongxiang, 2025-05-08:
Hi, sorry for the late response. The ROS2 demo for handeye calibration is under evaluation, since it involves other third party repositories and some of which have not supported ROS2 when we first created this repo.

As for the re-projection error, we have not yet done the evaluation. The calibration result may be affected by many factors and we are not expert at this (since we are just robot arm manufacturer). We will do the evaluation once we have the D435 camera (now borrowed by another project). You can follow the precautions we found online [here](https://github.com/xArm-Developer/xarm_ros?tab=readme-ov-file#722-precautions-for-hand-eye-calibration) which may be useful to improve the result.

Meanwhile, maybe you can ask for the error range from [easy-handeye](https://github.com/IFL-CAMP/easy_handeye) team or other forums where vision/camera calibration experts populate.