xArm-Developer/xarm_ros2 #128: MoveIt Servo stuck at "Command type has not been set, cannot accept input" even with valid JointJog / Twist commands

nico7093, 2025-05-04:
## Description

**Environment**

- ROS 2 Humble
- MoveIt 2 (Humble version)
- xarm_ros2 (humble branch)
- `xarm_moveit_servo` used in simulated mode (`xarm_moveit_servo_fake.launch.py`) and tested with both keyboard jog and manual ROS 2 topic publishers.

---

**Problem Summary**

When launching `xarm_moveit_servo_fake.launch.py` and using either:

✅ `xarm_keyboard_input` → keyboard jog  
✅ `ros2 topic pub` → manual JointJog or TwistStamped

The `servo_server` **rejects all input and never starts jogging**. The console keeps repeating this log:

```
[servo_server]: Command type has not been set, cannot accept input
```

Even after sending valid commands, e.g.:

```bash
ros2 topic pub /servo_server/delta_joint_cmds control_msgs/msg/JointJog "{joint_names: ['joint1'], displacements: [0.1], duration: 0.1}" -1
```

Or:

```bash
ros2 topic pub /servo_server/delta_twist_cmds geometry_msgs/msg/TwistStamped "{header: {frame_id: 'link_base'}, twist: {linear: {x: 0.05, y: 0.0, z: 0.0}, angular: {x: 0.0, y: 0.0, z: 0.0}}}" -1
```

**→ ALL commands are rejected with "Command type has not been set" and the robot does not move.**

---

## What was checked

✅ Topic names and message types → correct  
✅ Messages visible in `ros2 topic echo` → yes  
✅ `move_group_name` and planning group → valid and verified in RViz  
✅ No crashes → Servo runs normally otherwise  
✅ `expected_command_type` → stays UNSET forever

---

## Hypothesis (likely cause)

From reviewing `servo.cpp` and `command_interface.cpp`, it appears that:

- Servo starts in `CommandType::UNSET`
- Only when a valid command is processed does it call `setCommandType`
- However → current path with keyboard jog and manual topic pub does NOT trigger setting → DEADLOCK

Result:

- Servo stays in `CommandType::UNSET`
- All commands are rejected → nothing works → infinite "Command type has not been set" spam

---

## Additional info

If requested, I can provide:

- Full logs
- Launch commands
- A reproducible test workspace

---

Thank you very much for your help and consideration — this would really help simplify using MoveIt Servo for non-joystick / keyboard users!

---

# Labels:
`servo` `bug` `moveit_servo` `ros2` `keyboard_teleop` `command_type`

penglongxiang, 2025-05-08:
Hi @nico7093, thank you for the detailed feedback, we will look into it, are you using `jazzy` branch of this repository?

vimior, 2025-05-13:
@nico7093 
Hello, I looked through the moveit code, `command_type` was introduced from the `iron` version. I don't know how your `humble` branch can support `command_type`. If possible, I suggest you use the `jazzy` branch.

Saurabh0325, 2025-07-15:
will the repository for jazzy work in humble setup?
I am experiencing the same issue and could not find any solution