xArm-Developer/xarm_ros2 #133: xArm 5 vacuum gripper control in ROS2 Humble

denn-gubsky, 2025-07-17:
Vacuum gripper {xarm}/set_vacuum_gripper service does not work on real xArm 5 with mounted vacuum gripper in ROS2 Humble.

I have enabled xArm vacuum gripper services in xarm_api/config/xarm_user_params.yaml file and vacuum gripper services are visible in the service list:

```
$ ros2 service list | grep "xarm"
/xarm/clean_error
/xarm/clean_warn
/xarm/get_servo_angle
/xarm/get_vacuum_gripper
/xarm/motion_enable
/xarm/move_gohome
/xarm/set_mode
/xarm/set_position
/xarm/set_servo_angle
/xarm/set_servo_angle_j
/xarm/set_state
/xarm/set_vacuum_gripper

```
The /xarm/get_vacuum_cleaner service always responds with data == 0, regardless of the current vacuum gripper state:

```
$ ros2 service call /xarm/get_vacuum_gripper xarm_msgs/srv/GetInt16
requester: making request: xarm_msgs.srv.GetInt16_Request()

response:
xarm_msgs.srv.GetInt16_Response(ret=0, message='hardware_version=0, data=0', data=0)

```
If I try to change the vacuum gripper state with 

```
$ ros2 service call /xarm/set_vacuum_gripper xarm_msgs/srv/SetInt16 "{data: 1}"
waiting for service to become available...

```
It never responds. 

Is there a real support of the xArm 5 vacuum gripper in ROS2?

penglongxiang, 2025-07-18:
Hi @denn-gubsky in **xarm_ros2**, vacuum gripper service is not enabled by default. Please read the section [here](https://github.com/xArm-Developer/xarm_ros2?tab=readme-ov-file#54-xarm_api) to **enable the corresponding services** in configuration file.

In particular, you may create an `xarm_user_params.yaml` file in `xarm_api/config` folder, copy the content from `xarm_params.yaml` file and then do the edit, so that the change would not be overwritten by future `git pull`.

denn-gubsky, 2025-07-18:
@penglongxiang Thank you for your quick response and for your kind help. The `/xarm/set_vacuum_gripper` service with correct type now responds immediately and works correctly if I provide the `'hardware_version': 2` parameter: 

```
$ros2 service call /xarm/set_vacuum_gripper xarm_msgs/srv/VacuumGripperCtrl "{'on': true, 'hardware_version': 2}"
requester: making request: xarm_msgs.srv.VacuumGripperCtrl_Request(on=True, wait=False, timeout=3.0, delay_sec=0.0, sync=True, hardware_version=2)

response:
xarm_msgs.srv.VacuumGripperCtrl_Response(ret=0, message='hardware_version=2')

```
The `/xarm/get_vacuum_gripper` service returns correct hardware version, but vacuum gripper status is always 0.

```
ros2 service call /xarm/get_vacuum_gripper xarm_msgs/srv/GetInt16 
requester: making request: xarm_msgs.srv.GetInt16_Request()

response:
xarm_msgs.srv.GetInt16_Response(ret=0, message='hardware_version=2, data=0', data=0)
```
I will cache gripper status in my controller internally.

closed 2025-07-18 by denn-gubsky