xArm-Developer/xarm_ros2 #142: Error installing dependencies

alecdelcid, 2025-09-29:
Hi I'm getting this error when installing dependencies. I've been copy pasting each step so it should work. Can someone help me out?
/usr/bin/rosdep:6: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  from pkg_resources import load_entry_point
WARNING: ROS_PYTHON_VERSION is unset. Defaulting to 3
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
xarm_description: Cannot locate rosdep definition for [launch_ros]
xarm_moveit_servo: Cannot locate rosdep definition for [std_srvs]
xarm_gazebo: Cannot locate rosdep definition for [controller_manager]
mbot_demo: Cannot locate rosdep definition for [diff_drive_controller]
xarm_planner: Cannot locate rosdep definition for [moveit_msgs]
xarm_sdk: Cannot locate rosdep definition for [ament_cmake]
xarm_msgs: Cannot locate rosdep definition for [geometry_msgs]
xarm_api: Cannot locate rosdep definition for [geometry_msgs]
realsense_gazebo_plugin: Cannot locate rosdep definition for [rclcpp]
xarm_moveit_config: Cannot locate rosdep definition for [backward_ros]
xarm_controller: Cannot locate rosdep definition for [controller_manager_msgs]
uf_ros_lib: Cannot locate rosdep definition for [launch_ros]

closed 2025-09-29 by alecdelcid

rperdomo-research, 2026-02-09:
What was the fix for this? Having a similar issue with Jazzy

`ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
xarm_gazebo: Cannot locate rosdep definition for [sdformat14]`

Lau86, 2026-02-11:
Hi, I am currently using Ubuntu 24.04 with ROS 2 Jazzy and am encountering the same issue during the installation process. Have you managed to resolve it ?  If so, I would be very grateful if you could share the solution.

Thank you in advance!

```
ERROR: the following packages/stacks could not have their rosdep keys resolved to system dependencies:
xarm_gazebo: Cannot locate rosdep definition for [sdformat14]
```

Lau86, 2026-02-23:
Hi @rperdomo-research ,

Thank you very much for sharing this solution.

I followed your suggestion and it is working on my side as well. Everything builds and runs correctly.

Thanks again !